weewx/weewx #1019: Station Registry HTTP Error 429: TOO MANY REQUESTS

evilbunny2008, 2025-09-26:
I doubt most users will encounter this most of the time but when testing code and/or config changes this appears in logs every time WeeWX is restarted.

`weewxd[89180]: ERROR weewx.restx: StationRegistry: Failed to publish record 2025-09-26 12:45:00 AEST (1758854700): HTTP Error 429: TOO MANY REQUESTS`

So I thought before coming up with a PR I'd give my thoughts and ask for opinions on solving this minor issue.

My initial thoughts on stopping WeeWX even trying to publish to frequently is to have an entry in weewx.conf that tracks the time of the last successful attempt.

Maybe even a slight change server side as well, so that instead of rejecting, it also gives the time of the last successful registration which would already be known.

tkeffer, 2025-09-26:
The server knows the time of the last successful registration, then checks the new post against that time. It issues a `429` if the time is too short. However, there's no way for the server to know whether it's short because the user has decided to send a lot of requests, or because someone restarted the daemon. 

Of course, the client knows, but keeping track of the last post would require some client state, which I really want to avoid. I especially don't want `weewxd` modifying `weewx.conf`. It should be regarded as "read only."

The error occurs only when you restart the daemon, at which point a user should not be surprised by the error. I don't see it as a problem worth solving.

I should add that lots of other upload sites do something similar.

Correct me if I'm missing something subtle. It's certainly possible!

evilbunny2008, 2025-09-27:
Initially I considered modifying the timestamp on a blank file somewhere, but then thought the weewx.conf file already exists, but if that's out then maybe a blank file somewhere then.